kitsuned/PotatoNV #5: Kirin 970?

Stoobacca, 2020-08-02:
Hey, sorry not an issue.. but, do you think this can work for Kirin 970 devices, and does it require testpoint? Happy to test on 970! Cheers

YijunYuan, 2020-08-05:
I don't think this program supports Kirin 970 now since there are only 3 folders in the `bootloaders` fold of the release: `hisi659a`, `hisi659b`, `hisi960`. However, it would be nice, as we hope, to have the support for `hisi970`. Hope there is no breaking change.

mashed-potatoes, 2020-08-07:
Unfortunately, fastboot on Kirin 970/980/710/810 and newer ignores USRKEY & FBLOCK properties in NVME.
So, PotatoNV theoretically can flash the bootloader and update NVME on new SoCs, but this will not affect the bootloader status.

closed 2020-08-09 by mashed-potatoes

Slluxx, 2022-02-09:
> Unfortunately, fastboot on Kirin 970/980/710/810 and newer ignores USRKEY & FBLOCK properties in NVME. So, PotatoNV theoretically can flash the bootloader and update NVME on new SoCs, but this will not affect the bootloader status.

could you display the current USRKEY on unsupported SoC's? Its a sha256 string, right? Is there an easy way to display that without a lot of this low-level knowledge?