kitsuned/PotatoNV #6: Can you please tell , how to use this files , much thanks in advance :)

titanx980ti, 2020-09-02:
it would be better , if it has a detailed instructions on how to open this file

micbanand, 2020-09-02:
https://github.com/mashed-potatoes/PotatoNV/releases
there are 2 links in xda thread. Think you miss the one I linked here

micbanand, 2020-09-02:
> it would be better , if it has a detailed instructions on how to open this file

https://github.com/mashed-potatoes/PotatoNV/releases
there are 2 links in xda thread. Think you miss the one I linked here

titanx980ti, 2020-09-02:
i saw that , for some reason my phone is detected only in fastboot , later gives a output saying , FP locked !, will this help in getting the boot loader code ?

micbanand, 2020-09-02:
read up on oem lock. (is in dev options) and basic adb. and I will recommend not logged in to google account and no use of code. 
Out

mashed-potatoes, 2020-09-02:
Please follow the guide: https://github.com/mashed-potatoes/PotatoNV/wiki/How-to-use-PotatoNV

titanx980ti, 2020-09-02:
> 
> 
> read up on oem lock. (is in dev options) and basic adb. and I will recommend not logged in to google account and no use of code.
> Out

nope , i have not logged in with my google account , My FRP is unlocked , But my phone is locked ! , that is boot loader ,where i need oem unlock code to unlock it.

mashed-potatoes, 2020-09-09:
Since I have not heard back from you, I'm closing this issue.

closed 2020-09-09 by mashed-potatoes