kitsuned/PotatoNV #8: Kickass!

insaneKane, 2020-09-26:
I know thats not what the Issues are ment for but this is an incredible tool.
I tried for quite a long time to unlock my Huawei P Smart and finally this tool is what worked.

ironOmega12, 2020-09-28:
Hey man, could you help me use this? I'm bad at programming but I tried to unlock my bootloader for so long..

ironOmega12, 2020-09-28:
> I know thats not what the Issues are ment for but this is an incredible tool.
> I tried for quite a long time to unlock my Huawei P Smart and finally this tool is what worked.
> 
> So thank you a ton :!

plz

mashed-potatoes, 2020-11-05:
> I know thats not what the Issues are ment for but this is an incredible tool.
> I tried for quite a long time to unlock my Huawei P Smart and finally this tool is what worked.
> 
> So thank you a ton :!

You're welcome!
If you found this tool helpful and you want to help the community, [consider donating to #niceguys](https://github.com/mashed-potatoes/PotatoNV/issues/10#issuecomment-722551274).

Sincerely,
Andrey

closed 2020-11-05 by mashed-potatoes