kitsuned/PotatoNV #10: Question about bootloaders

jayzsolt, 2020-11-03:
I'm glad I found your great tool, it's actually praised well on XDA Developers portal: https://forum.xda-developers.com/p10/how-to/july-2020-unlock-bootloader-huawei-t4135351

My question is, where do you get the bootloader binary from? I'm looking for one of Kirin 659 (P20 Lite), searched through the internet, even on shady chinese sites, could not find a fitting one... Any hint would be much appreciated! Good luck for future successes, or shall I say, Желаю вам всего хорошего !

jayzsolt, 2020-11-04:
Solved the issue, the problem was, I didn't compile your code properly. Downloaded a pre-compiled version from 4PDA and worked like a charm.

I'd suggest to put up a donation option to your website or in the software, happy to give you some credits for saving my ass :)

closed 2020-11-04 by jayzsolt

mashed-potatoes, 2020-11-05:
Glad you have it resolved. You can find the compiled PotatoNV on the ["Releases" page](https://github.com/mashed-potatoes/PotatoNV/releases). _(tbh I should add a link to this page in README)_

>_I'd suggest to put up a donation option to your website or in the software, happy to give you some credits for saving my ass :)_

If you want to help people who actively work and research Huawei consider donating to them:

- ETH: 0x4C7c645214F63ce9E0a6Ba5D388C9Ffb40c85B8f
- BTC: 1DFKpnqwWv1NCPM3NhW5DGrjuah6agLw7M
- PayPal: https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=HNL4MYLLPEVVC&source=url

Regards.
Andrey