kitsuned/PotatoNV #15: Read bootloader code instead of overwriting it

Microeinstein, 2020-12-14:
Hi - first of all, thank you for your nice tool.

I noticed you require the user to change their bootloader code, by writing the new value under `USRKEY` in the form

    number -> ASCII (not null-term.) -> SHA256

On my phone (P8 Lite 2015) this is not required, because it is **also written in plaintext** under `WVLOCK`, thus it can be recovered with `getvar:nve:WVLOCK`.
I don't know if this is valid for every other phone, but I think you can easily add some checks for it 👍🏻

mashed-potatoes, 2021-02-05:
It will be implemented in the new release, thanks for the idea.

2BDia, 2021-03-19:
Hi,
I also have a P8 Lite (ale l21) and I was wondering where I could find this WLOCK file to have the bootloader code.

Thanks a lot in advance

Microeinstein, 2021-03-19:
@2BDia
You must access the `nvme` partition; it could be done in two ways:
(_**DISCLAMER**: your phone, your responsibility._)

## Fastboot
### Requirements
* `FBLOCK` = `0`
    * In this mode you can execute any hidden internal command
    * It can be set by
        1. Flashing a stock bootloader in RAM via IDT tool with the testpoint method
        2. Executing `fastboot getvar 'nve:$FBLOCK@00'`
        3. Rebooting normally into fastboot
        * or with root, by manually changing the values into that partition with a special `sed` command
### Execution
1. Run `fastboot getvar 'nve:WVLOCK'`

## Root
### Execution
1. Special `grep` command over that partition

---

At the end you can easily unlock your bootloader with `fastboot oem unlock CODE`.
(yes, I know that to have a rooted phone you must have the bootloader already unlocked; I added those procedures just to be comprehensive)

2BDia, 2021-03-19:
Thanks for your answer!
So basically I either have to find a way to set FBLOCK to 0 (looking into it rn) or root my phone somehow
I was wondering if downgrading to Android 5.0, rooting and getting the bootloader code, then upgrading back to Android 6.0 would work.

mashed-potatoes, 2021-03-19:
> Hi,
> I also have a P8 Lite (ale l21) and I was wondering where I could find this WLOCK file to have the bootloader code.
> 
> Thanks a lot in advance

See https://github.com/mashed-potatoes/PotatoNV/issues/39#issuecomment-803104732

mashed-potatoes, 2021-03-30:
Done!

closed 2021-03-30 by mashed-potatoes