kitsuned/PotatoNV #19: Huawei P20 Lite

Timusss2378, 2021-01-26:
FBLOCK - disable issue
- Model: ANE-LX1
- Build number: ANE-LX1 9.1.0.352(C10E6R1P6)
- FBLOCK state: unlocked
--> Updating NVME
- Writing FBLOCK state
**Failed to set prop:**

affineer, 2021-02-25:
> FBLOCK - disable issue
> 
> * Model: ANE-LX1
> * Build number: ANE-LX1 9.1.0.352(C10E6R1P6)
> * FBLOCK state: unlocked
>   --> Updating NVME
> * Writing FBLOCK state
>   **Failed to set prop:**

Could you find solution to this problem, I am in the same boat. :(

mashed-potatoes, 2021-03-19:
See https://github.com/mashed-potatoes/PotatoNV/issues/39#issuecomment-803104732

Albezan, 2021-03-20:
I have the same issue:

PotatoNV v2.0.1
User manual: https://kutt.it/pnv-en
--> Flashing bootloader
Verifying images...
Verification passed!
Uploading hisi659a bootloader
- xloader
- fastboot
Bootloader uploaded
Waiting for any device...
--> Reading information
Connecting to fastboot device...
- Serial number: xxxxxxxxxxxxxxxx
- Board ID: DU4VYH1868047045
- Model: ANE-LX1
- Build number: ANE-LX1 9.1.0.376(C432E7R1P7)
- FBLOCK state: unlocked
--> Updating NVME
- Writing FBLOCK state
**Failed to set prop:**
How can i solve it???
The smartphone is P20 Lite and  it boot only in fastboot mode.

BudDeuce, 2021-05-02:
@Albezan Did you have it in test point mode?

BudDeuce, 2021-05-03:
@Timusss2378 Did you have it in test point mode?

Albezan, 2021-05-17:
Yes I did it.
I think the smartphone have a problem to emmc .....

Il

> @Albezan [1] Did you have it
in test point mode? 
mentioned.
unsubscribe [3].

BudDeuce, 2021-05-17:
[image]

Oh, alright then. Here's the test point in case you need it.

@Albezan