kitsuned/PotatoNV #20: How to use PotatoNV

ThomasHeinrichSchmidt, 2021-01-30:
# How to use PotatoNV

## Unlocking the boot loader 
_DISCLAIMER: You will void your device warranty. I have successfully done this guide but I am not responsible for any malfunctions or any damage to the device. Please read carefully every single step and everything will be successful._ 

Unlocking puts your device in a potentially unsafe state in terms of vulnerability to malware.
Incidentally, unlocking will not change the installed software and will not damage your data. 
Back up anything you may need to be safe. 

After unlocking the device will show a message before booting but will otherwise start normally as before. 
[image]

### Install ADB and HUAWEI USB COM drivers
   After ADB installation, copy 5 files to PotatoNV-next.exe bin directory
``
      adb.exe         
         AdbWinUsbApi.dll
         AdbWinApi.dll
      fastboot.exe
         libwinpthread-1.dll
``
Install HUAWEI USB COM drivers (using "Huawei drivers testpoint.rar" from https://files.dc-unlocker.com/share.html?v=share/18B15B9D02C945A79B1967234CECB423).
This installs without any problems, the drivers are signed by Huawei, confirmed by VirusTotal and Windows.
Open up Device Manager to check that under ports (COM & LPT) connection changes from 'USB SER' to 'USB (COM)'
Uninstall the HUAWEI drivers when finished or else phone won't be recognized properly by PC later!
### Remove the cover from the device: 
Heat the cover on the back of the phone with a hairdryer and carefully peel cover along the small gap using a plastic card or guitar pick 
[image]

### Find where the testpoint is located. 
Sample search phrase "HUAWEI P8 Lite 2017 PRA LA1 testpoint" will result in several hits with with suitable photos to show you the location of the testpoint. In my case it is enough to open the back cover of the phone, you do not need to remove any cables or covers.

### ❋ Init 
Turn off the device and unplug the USB cable

### Short testpoint and iron shield (ground) on the motherboard
Solder a piece of flexible cable to a rigid cable, clamp some wires of the flexible cable to the external part of the UBS connector by connecting it to the phone.
[image]
You should use a multimeter or resistance meter to check if there is a conductive connection with the outside of the USB connector on the PC side, i.e. if your cable connection to ground is good, 
[image] 
[image]
   Touch the test point with the other stiff end of the wire 
### Connect the Phone to the PC 
using the USB connector, while still touching the test point with the wire 
### Now turn on the phone 
by pushing the power button for 10 seconds, while keeping the cable connected at both ends.
Make sure there’s nothing on the device’s screen. If a charging animation appears go to step **❋ Init**

_I had problems to watch the (front) screen while holding the cable connected on the background of the device, so I did not see what happened (maybe that's the reason why my first attempt failed). But I just tried again from step ❋ Init and it worked the second time._

### Launch PotatoNV 
and select "HUAWEI USB COM 1.0" in the device list (was already automatically chosen)
Select bootloader - see table to help you make a choice. If the device is not in the table, then just try different bootloaders (this is safe).
For my P8 Lite 2017 (PRA-LX1) PotatoNV chose "Kirin 659(A)" by default what was OK.
Fields SN and BOARDID are optional, just leave them blank
Unlock code - this is the future bootloader code, you can change it to any string, the code must contain 16 characters
Do not forget to write down the bootloader unlock code PatatoNV generates (you may need it while trying to install the new system or using eRecovery
### Press the Start button on PotatoNV.
PotatoNV shows:
```
         PotatoNV v2.0.1
            User manual: https://kutt.it/pnv-en
            --> Flashing bootloader
            Verifying images...
            Verification passed!
            Uploading hisi659a bootloader
            - xloader
            - fastboot
            Bootloader uploaded
            Waiting for any device...
            --> Reading information
            Connecting to fastboot device...
            - Serial number: 1234567890123456
            - Board ID: ABCD567890123456
            - Model: PRA-LX1
            - Build number: PRA-LX1 8.0.0.410(C432)
            - FBLOCK state: unlocked
            --> Updating NVME
            - Writing FBLOCK state
            - Writing User key
            Update done!
            Rebooting...
```   
[image]

If everything goes well, the phone will reboot and show the following text: "Your device has been unlocked, ..." during boot, see first picture "PHONE Booting unlocked" at top.
### Verify that everything worked out
first you need to enable USB-debugging in the development options of your device then type in command window:
```
      C:\> adb devices
         List of devices attached
         1234567890123456         device
      C:\> fastboot oem get-bootinfo
         < waiting for any device >
         ^C        -- does not work 
      C:\> adb reboot bootloader
                      -- phone shows: PHONE Unlocked (red) / FRP Unlock (red)   [FRP=Factory Reset Protection]
      C:\>fastboot oem get-bootinfo
         ...
         **(bootloader)  unlocked**   
         OKAY [  0.011s]
         finished. total time: 0.013s
      C:\> fastboot reboot
         rebooting...
         finished. total time: 0.026s
``` 
[image]

### Uninstall HUAWEI USB COM drivers 
To be able to access the phone via USB (to copy files, run ADB etc.) you need to uninstall the HUAWEI USB COM drivers using "DriverUninstall.exe".
Once the uninstall is finished it shows a strange "Restart" message box 
[image]

If you press OK, your PC will restart (what is probably necessary to actually remove the driver).

mishailovic, 2021-01-30:
His ideas will ALWAYS be relevant

dnitto, 2021-02-21:
Hello good Morning
I have the huawei mate 9 mha l09 I should connect my phone to the pc open the program and start just that or make that connection that you see with the usb cable.

closed 2021-02-22 by mashed-potatoes