kitsuned/PotatoNV #26: Honor 6x - Unlock worked

Riesi, 2021-02-24:
Thank you so much! I never thought that someone would come up with a method to unlock it. <3
```
PotatoNV v2.0.1
User manual: https://kutt.it/pnv-en
--> Flashing bootloader
Verifying images...
Verification passed!
Uploading hisi659a bootloader
- xloader
- fastboot
Bootloader uploaded
Waiting for any device...
--> Reading information
Connecting to fastboot device...
- Serial number: <redacted>
- Board ID: <redacted>
- Model: BLN-L21
- Build number: BLN-L21C432B382
- FBLOCK state: unlocked
--> Updating NVME
- Writing FBLOCK state
- Writing User key
Update done!
Rebooting...
Bootloader unlock code: 1111111111111111
```

risturiz, 2021-02-25:
Hi, what bootloader is "hisi659a" ? Thanks.

Riesi, 2021-02-25:
Kirin 659 (A)

mashed-potatoes, 2021-02-27:
I'm glad to hear it 👍

closed 2021-02-27 by mashed-potatoes

risturiz, 2021-03-07:
For the record BLN-L24 works too...Thanks.

bliu13, 2021-08-25:
Can confirm the unlock worked with BLN-L24. If anyone else was curious, the device bootloader is Kirin 65x (A).