kitsuned/PotatoNV #36: How to (re)lock bootloader again?

BurnyLlama, 2021-02-27:
I don't know if this should be in "issues", but I'll give it a try.

I have a Huawei P9 Lite (VNS-L31). It couldn't boot before (stuck on the Huawei Logo). I could only access fastboot, but it had a locked bootloader, so I couldn't do anything.

I found this tool (luckily - I am very grateful for it), but now I have trouble locking the bootloader. I want to return to stock and passing safety net, etc.

How would I (re)lock the bootloader? Or is it impossible because the device has been opened?

Thanks in advance!

mashed-potatoes, 2021-02-27:
Hi,
try these commands:

```bash
fastboot oem hwdog certify set 1
fastboot reboot
```

BurnyLlama, 2021-02-27:
OMG!

> Hi,
> try these commands:
> 
> ```shell
> fastboot oem hwdog certify set 1
> fastboot reboot
> ```

OMG! It seems to have worked! Thank you very much! :D <3

closed 2021-02-27 by BurnyLlama