kitsuned/PotatoNV #39: Huawei P8lite ALE-L21 (2015)

juanmartin, 2021-03-04:
Hi, any chance for this one to work?
I just tried but got stuck with xloader, it said "write time out".
Also I realised my phone is Kirin620, which is not available as a bootloader so I guess it would fail anyway, right?

It is actually my mom's phone, I'm just trying to avoid her spending a ton of money on a new phone when there should be a way of getting out of a sudden bootloop!

Thanks!

juanmartin, 2021-03-05:
Bump! I added some detail to the post.

Thespartann, 2021-03-13:
**I also tried unlocking the bootloader of a Huawei P8 Lite 2015 (ALE-L21) with PotatoNV with the same result.**

Unlocking on the kirin620 is possible trough USB COM 1.0 mode with other proprietary tools, should be easily added to PotatoNV.
For the P8 lite the bootloader code can also be read, doesn't even have to be replaced.

If you just want to unlock right now wihout any tools you can downgrade to lollipop EMUI firmware, root and then read the code in adb shell like this:
### su -c "grep -m1 -aoE 'WVLOCK.{14}[0-9]{16}' /dev/block/mmcblk0p7 |grep -aoE '[0-9]{16}'"

juanmartin, 2021-03-15:
Thank you @Thespartann, I feel less alone now.
I tried putting a Lollipop image in a sd card but it wouln't boot, how can I downgrade?

mashed-potatoes, 2021-03-19:
Okay, try this: https://ci.appveyor.com/api/buildjobs/wgsuob9yx6f537os/artifacts/PotatoNV-next%2Fbin%2FPotatoNV-appveyor.zip

---

__In any case, please provide logs:__

[image]

arturoguzmanp, 2021-03-20:
Hi, I tried the version (2.1.0) from the link but it doesn't work, don't know if a file is missing or there's a runtime error.

here's the log:

2021-03-20 13:01:11.494 | INFO     | PotatoNV v2.1.0
2021-03-20 13:01:11.509 | INFO     | User manual: https://kutt.it/pnv-en
2021-03-20 13:01:45.771 | DEBUG    | DownloadVCOM mode: COM5: HUAWEI USB COM 1.0
2021-03-20 13:01:54.637 | INFO     | PotatoNV v2.1.0
2021-03-20 13:01:54.637 | INFO     | User manual: https://kutt.it/pnv-en
2021-03-20 13:01:54.652 | INFO     | Verifying images...
2021-03-20 13:01:54.668 | DEBUG    | Opening COM5...
2021-03-20 13:01:54.684 | ERROR    | No se puede cargar el archivo o ensamblado 'System.IO.Ports, Version=4.0.1.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' ni una de sus dependencias. El sistema no puede encontrar el archivo especificado.
2021-03-20 13:01:54.699 | DEBUG    | en Potato.ImageFlasher.ImageFlasher.Open(String portName)
    en PotatoNV_next.Core.FlashBootloader(Bootloader bootloader, String port) en C:\projects\potatonv\PotatoNV-next\Core.cs:línea 41
    en PotatoNV_next.Core.Worker_DoWork(Object sender, DoWorkEventArgs e) en C:\projects\potatonv\PotatoNV-next\Core.cs:línea 181

mashed-potatoes, 2021-03-20:
Thanks for the response, please wait 10-20 mins.

mashed-potatoes, 2021-03-20:
@arturoguzmanp, try this:
[v2.1.0-test2.zip](https://github.com/mashed-potatoes/PotatoNV/files/6176311/v2.1.0-test2.zip)

arturoguzmanp, 2021-03-20:
Thank you, it runs now! Although I am having a timeout error, the device I am trying to unlock is using Kirin 620. Here is the log:

2021-03-20 13:52:36.996 | INFO     | PotatoNV v2.1.0
2021-03-20 13:52:37.002 | INFO     | User manual: https://kutt.it/pnv-en
2021-03-20 13:53:23.312 | DEBUG    | DownloadVCOM mode: COM5: HUAWEI USB COM 1.0
2021-03-20 13:53:49.835 | INFO     | PotatoNV v2.1.0
2021-03-20 13:53:49.835 | INFO     | User manual: https://kutt.it/pnv-en
2021-03-20 13:53:49.835 | INFO     | Verifying images...
2021-03-20 13:53:49.851 | DEBUG    | Opening COM5...
2021-03-20 13:53:49.851 | INFO     | Uploading hisi620...
2021-03-20 13:53:49.851 | INFO     | - xloader
2021-03-20 13:53:51.035 | INFO     | - fastboot
2021-03-20 13:53:52.051 | ERROR    | La escritura superó el tiempo de espera.
2021-03-20 13:53:52.051 | DEBUG    | en System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
    en System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
    en Potato.ImageFlasher.ImageFlasher.SendFrame(Byte[] data) en C:\projects\potatonv\Potato.ImageFlasher\Potato.ImageFlasher\ImageFlasher.cs:línea 122
    en Potato.ImageFlasher.ImageFlasher.SendHeadFrame(Int32 length, Int32 address) en C:\projects\potatonv\Potato.ImageFlasher\Potato.ImageFlasher\ImageFlasher.cs:línea 82
    en Potato.ImageFlasher.ImageFlasher.Write(String path, Int32 address, Action`1 reportProgress) en C:\projects\potatonv\Potato.ImageFlasher\Potato.ImageFlasher\ImageFlasher.cs:línea 48
    en PotatoNV_next.Core.FlashBootloader(Bootloader bootloader, String port) en C:\projects\potatonv\PotatoNV-next\Core.cs:línea 51
    en PotatoNV_next.Core.Worker_DoWork(Object sender, DoWorkEventArgs e) en C:\projects\potatonv\PotatoNV-next\Core.cs:línea 181
2021-03-20 13:54:10.717 | DEBUG    | DownloadVCOM mode: COM5: HUAWEI USB COM 1.0

mashed-potatoes, 2021-03-20:
That's weird. Could you try [this](https://github.com/mashed-potatoes/PotatoNV/files/6176335/ALE-L21_HL3ALICEMAA001079_Board_Software_05021SJN_HMT.zip) package with [HiSilicon IDT](https://androidfilehost.com/?fid=4349826312261681288)?

arturoguzmanp, 2021-03-20:
I've tried but IDT says 'ptable.img' doesn't exist in the ale-l21..../fastbootimage folder.

juanmartin, 2021-03-20:
> That's weird. Could you try [this](https://github.com/mashed-potatoes/PotatoNV/files/6176335/ALE-L21_HL3ALICEMAA001079_Board_Software_05021SJN_HMT.zip) package with [HiSilicon IDT](https://androidfilehost.com/?fid=4349826312261681288)?

This build worked for my ALE-L21 (2015)! Thank you so much.
I didn't have to use any IDT (I dunno what it is though). I have ADB, HiSuite (useless..) and Testpoint Drivers installed. The device was recognized after connecting and restarting with grounded testpoint, then just pressed Start!

```
PotatoNV v2.1.0
User manual: https://kutt.it/pnv-en
Verifying images...
Uploading hisi620...
- xloader
- fastboot
Waiting for any device...
Connecting...
Serial number: XXXXXXXXXXXXXX
Board ID: XXXXXXXXXXXXX
Model: :ALE-L21
Build number: ALE-L21C55B587
FBLOCK state: locked
FBLOCK is locked!
```

The phone entered FASTBOOT&RESCUE MODE immediately  when "Waiting for any device" and now it says `PHONE Unlocked`.
I'm worried about that last line `FBLOCK is locked!` though... what does that mean?
I guess I can now install any recovery and system I want? I'll give that some research and a try, and hopefully revive the device.

arturoguzmanp, 2021-03-20:
Good it worked on your phone! guess the issue could be my phone then. (CAM-L03)

juanmartin, 2021-03-20:
I know it may have nothing to do but since I'm talking about the same device I guess I would share this.
Immediately after unlocking the bootloader I tried installing a TWRP and it hard bricked the phone... is there anything I can do with testpoint and this tool in order to save it.. again?

mashed-potatoes, 2021-03-21:
> I know it may have nothing to do but since I'm talking about the same device I guess I would share this.
> Immediately after unlocking the bootloader I tried installing a TWRP and it hard bricked the phone... is there anything I can do with testpoint and this tool in order to save it.. again?

Please, run PotatoNV again, after getting 'FBLOCK is locked' run the following commands and copy the results:

```
fastboot oem lock-state info
fastboot oem get-lockstate
fastboot oem backdoor info
```

mashed-potatoes, 2021-03-22:
@juanmartin @arturoguzmanp I think it should work now.
[v2.1.0-test3.zip](https://github.com/mashed-potatoes/PotatoNV/files/6180212/v2.1.0-test3.zip)

Thespartann, 2021-03-22:
> I know it may have nothing to do but since I'm talking about the same device I guess I would share this.
> Immediately after unlocking the bootloader I tried installing a TWRP and it hard bricked the phone... is there anything I can do with testpoint and this tool in order to save it.. again?

**Yes! There is a way :)**

I actually looked into it and hard bricked my P8 lite then i fixed it.

So first question is how did you flash twrp?

Basically to revive your phone first it is very important to do a backup of all partitions then stock firmware can be flashed via Image Download Tool or other tools.

Here is the link for board firmware: https://drive.google.com/file/d/1zlRgVlT7EUlrM8QA-Qrdqn_LlGUJ-tVr

**It is crucial to backup firmware partitions as from my experience flashing board firmware actually wipes IMEI and other data.**

To not spam here you can message me on telegram @Thespartann or in the Huawei P8 Lite group @huawei_p8lite (https://t.me/huawei_p8lite) for unbricking instructions, good luck :)

Also thanks @mashed-potatoes for doing test builds for Kirin 620, i would test but i just got my P8 Lite back togheter :D

- [ ] I'm curious about using hisilicon loader to backup/flash partitions, for now i been only able to do it with MCT Dongle

arturoguzmanp, 2021-03-23:
> @juanmartin @arturoguzmanp I think it should work now.
> [v2.1.0-test3.zip](https://github.com/mashed-potatoes/PotatoNV/files/6180212/v2.1.0-test3.zip)

thanks @mashed-potatoes, sadly it is still timing out in the CAM-L03.

TBM13, 2021-04-27:
I tried to unlock my Huawei P8Lite (ALE-L23) with the version 2.2.0 of this tool. I got the same result as @juanmartin
This is the log:

> 2021-04-27 20:33:02.786 | INFO     | PotatoNV v2.2.0
2021-04-27 20:33:02.787 | INFO     | User manual: https://kutt.it/pnv-en
2021-04-27 20:33:03.052 | DEBUG    | DownloadVCOM mode: COM4: HUAWEI USB COM 1.0
2021-04-27 20:33:07.138 | INFO     | PotatoNV v2.2.0
2021-04-27 20:33:07.141 | INFO     | User manual: https://kutt.it/pnv-en
2021-04-27 20:33:07.145 | INFO     | Verifying images...
2021-04-27 20:33:07.153 | DEBUG    | Opening COM4...
2021-04-27 20:33:07.156 | INFO     | Uploading hisi620...
2021-04-27 20:33:07.156 | INFO     | - xloader
2021-04-27 20:33:08.325 | INFO     | - fastboot
2021-04-27 20:33:14.679 | INFO     | Waiting for any device...
2021-04-27 20:33:17.729 | INFO     | Connecting...
2021-04-27 20:33:17.734 | DEBUG    | Fastboot mode: XXXXXXXXXXXXXX
2021-04-27 20:33:17.738 | INFO     | Serial number: XXXXXXXXXXXXXX
2021-04-27 20:33:17.748 | INFO     | Board ID: XXXXXXXXXXXXXX
2021-04-27 20:33:17.821 | INFO     | Model: :ALE-L23
2021-04-27 20:33:17.891 | INFO     | Build number: ALE-L23C605B534
2021-04-27 20:33:17.891 | INFO     | FBLOCK state: locked
2021-04-27 20:33:17.892 | DEBUG    | response: FAILinvalid command                                             
2021-04-27 20:33:17.892 | ERROR    | *** FBLOCK is locked! ***
2021-04-27 20:33:17.901 | DEBUG    | en PotatoNV_next.Core.ReadInfo()
    en PotatoNV_next.Core.Worker_DoWork(Object sender, DoWorkEventArgs e)

TBM13, 2021-04-27:
It seems i got a similar result with the v2.1.0-test3 version. Here's the log:

> 2021-04-27 20:46:48.250 | INFO     | PotatoNV v2.1.0
2021-04-27 20:46:48.251 | INFO     | User manual: https://kutt.it/pnv-en
2021-04-27 20:46:48.505 | DEBUG    | DownloadVCOM mode: COM4: HUAWEI USB COM 1.0
2021-04-27 20:46:57.487 | INFO     | PotatoNV v2.1.0
2021-04-27 20:46:57.487 | INFO     | User manual: https://kutt.it/pnv-en
2021-04-27 20:46:57.490 | INFO     | Verifying images...
2021-04-27 20:46:57.513 | DEBUG    | Opening COM4...
2021-04-27 20:46:57.516 | INFO     | Uploading hisi620...
2021-04-27 20:46:57.517 | INFO     | - xloader
2021-04-27 20:46:58.689 | INFO     | - fastboot
2021-04-27 20:47:05.054 | INFO     | Waiting for any device...
2021-04-27 20:47:07.919 | DEBUG    | Fastboot mode: XXXXXXXXXXXXXX
2021-04-27 20:47:08.151 | INFO     | Connecting...
2021-04-27 20:47:08.163 | INFO     | Serial number: XXXXXXXXXXXXXX
2021-04-27 20:47:08.173 | INFO     | Board ID: XXXXXXXXXXXXXX
2021-04-27 20:47:08.248 | INFO     | Model: :ALE-L23
2021-04-27 20:47:08.440 | INFO     | Build number: ALE-L23C605B534
2021-04-27 20:47:08.441 | INFO     | FBLOCK state: locked
2021-04-27 20:47:08.442 | DEBUG    | response: FAILinvalid command                                             
2021-04-27 20:47:08.442 | ERROR    | FBLOCK is locked!
2021-04-27 20:47:08.443 | INFO     | Writing FBLOCK...
2021-04-27 20:47:08.513 | DEBUG    | response: OKAYset nv ok                                                   
2021-04-27 20:47:08.513 | INFO     | Writing USRKEY...
2021-04-27 20:47:08.548 | DEBUG    | response: OKAYset nv ok                                                   
2021-04-27 20:47:08.548 | INFO     | Rebooting...
2021-04-27 20:47:08.569 | INFO     | New bootloader unlock code: 2PJJEOJLCDWT2MQA
2021-04-27 20:47:51.157 | DEBUG    | Fastboot mode: XXXXXXXXXXXXXX

TBM13, 2021-04-28:
`fastboot oem lock-state info` returns `invalid command`
`fastboot oem get-lockstate` returns `(bootloader) locked`
`fastboot oem backdoor info` returns 
> (bootloader)  FB LockState: UNLOCKED
> (bootloader)  USER LockState: LOCKED

`fastboot oem get-bootinfo` returns `(bootloader)  unlocked`

TBM13, 2021-04-28:
After overriding [this](https://github.com/mashed-potatoes/PotatoNV/blob/master/PotatoNV-next/Core.cs#L87) to be true, i was able to successfully unlock the bootloader.

jcbyrne1990, 2021-04-28:
I'm able to unlock bootloader via huawei com 1 first( if unsuccessful trigger with board software even if unsuceessful to flash)
fastboot triggered with "mct tool" ,a free tech tool (fastboot says unlocked this time)
then run with PotatoNV-next-v2.1.0-x86 (get unlock code)
this time reunlock via fastboot oem unlock (16digits code)     PotatoNV provided code

BudDeuce, 2021-05-03:
@juanmartin Do you have ADB installed?

juanmartin, 2021-05-03:
> @juanmartin Do you have ADB installed?

Yes, of course. I have it in `C:\adb\adb.exe`

Sorry I haven't been able to test anything again yet, will try to do these days. 

@Thespartann really helped me out, we even tried unbricking the phone remotely with mct tools but I think I screwed something up when I flashed TWRP. Maybe I used the wrong image and now the phone is more confused than me lol.

BudDeuce, 2021-05-03:
Where did you get your TWRP from?

juanmartin, 2021-05-03:
from [here](https://forum.xda-developers.com/t/huawei-p8-lite-install-twrp-recovery-v2-v3-0-0-v3-0-2-full-system-backup.3597170/)

BudDeuce, 2021-05-03:
@juanmartin Here is the official TWRP page: https://dl.twrp.me/prague/twrp-3.4.0-0-prague.img

Can you still get into fastboot mode?

plarfman, 2021-05-03:
@BudDeuce that's for PRA 2017 version, will NOT boot on ALE

BudDeuce, 2021-05-03:
@plarfman Whoops, your right.

My mistake.

BudDeuce, 2021-05-03:
[TWRP.3.2.3.ALE.zip](https://github.com/mashed-potatoes/PotatoNV/files/6416371/TWRP.3.2.3.ALE.zip)

I found another TWRP for your system.

juanmartin, 2021-05-03:
I can only get into testpoint mode u.u

BudDeuce, 2021-05-03:
I believe you will need a tool like DC Phoenix or HCU Client to help

plarfman, 2021-05-03:
@juanmartin how did you get it like that?

juanmartin, 2021-05-03:
> I believe you will need a tool like DC Phoenix or HCU Client to help

Yes  I tried MCT pro but cannot get it to backup my partitions, it timeouts. I think this is needed to keep IMEI and some other info right?

> @juanmartin how did you get it like that?

Flashing TWRP from the link I mentioned above.

BudDeuce, 2021-05-03:
@juanmartin I know Andrey Smirnoff has mentioned DC Phoenix as being a great tool for his needs.

juanmartin, 2021-05-03:
I was trying to avoid paying for this stuff... mainly because currency in my country is **** so it becomes an expensive solution for something that I don't even know whether it's going to work.

BudDeuce, 2021-05-03:
Looks like you'll need to replace your motherboard since that's where your partitions are stored.

mashed-potatoes, 2021-08-01:
Alice works now.

closed 2021-08-01 by mashed-potatoes

vlnahp, 2021-08-05:
Hi 
I have downloaded a xml format stock firmware for my p8 lite. But the xml file is invalid while using IDT tool v2.
Does anybody have a tested sample? I only need the xml file.