kitsuned/PotatoNV #43: Uploading bootloader times out

2BDia, 2021-03-17:
Hi,

I'm trying to unlock my Huawei P8 lite (ALE-L21) but I can't get past the "Uploading hisi659a bootloader" step (see screen)
[image]
It's probably due to ADB but I have no idea how to use it and install it, if someone would be willing to help. Ports are showing up with COM so it should be ok (?)
[image]

Thanks a lot in advance.

2BDia, 2021-03-17:
Little update : when the phone is on like usual, the device shows up in the list of connected adb devices, however I don't get anything when booting the phone with the testing point method
[image]

Again, thanks in advance :)

mashed-potatoes, 2021-03-20:
https://github.com/mashed-potatoes/PotatoNV/issues/39#issuecomment-803104732

BudDeuce, 2021-05-03:
Use a newer version of PotatoNV.

BudDeuce, 2021-05-03:
@2BDia Use a newer version of PotatoNV.

closed 2021-08-01 by mashed-potatoes