kitsuned/PotatoNV #45: Kirin 955

Picoseconds, 2021-03-18:
kirin 955, use potatonv?

mashed-potatoes, 2021-03-19:
Just try different bootloaders

Picoseconds, 2021-03-20:
So possible?

mashed-potatoes, 2021-03-20:
Maybe

NetviperZ, 2021-03-27:
any update with this kirin 955 ?

mohahack, 2021-04-04:
I used a kirin950 bootloader on Huawei P9 (Kirin955) and it works fine

Picoseconds, 2021-04-05:
Really? Great!

Picoseconds, 2021-04-05:
If that's true, I will close this issue

closed 2021-04-05 by Picoseconds

Picoseconds, 2021-04-05:
@mohahack Where is the testpoint

NetviperZ, 2021-04-07:
I just unlocked my P9 a week ago and i can confirm that kirin950 bootloader working fine with my Huawei P9 (EVA-L19)

[image]

[image]

mashed-potatoes, 2021-04-07:
@NetviperZ, thank you!