kitsuned/PotatoNV #59: Does it work on Huawei P10 Lite?

MrBlood01, 2021-04-17:
Hello, I read that among the various devices there is the Huawei P10, could it work on Huawei P10 Lite with Kirin 658?

plarfman, 2021-04-19:
it does, one of the Kirin 65x bootloaders will work

MrBlood01, 2021-04-19:
@plarfman I tried to do it. But it does not work. That is, the program does not detect my smartphone. I installed the android drivers and obviously the adb tools. I have enabled USB Debugging. But it doesn't work anyway.

plarfman, 2021-04-20:
it's probably becuase you haven't installed the USB COM drivers it needs, https://files.dc-unlocker.com/share.html?v=share/18B15B9D02C945A79B1967234CECB423 here's the link

MrBlood01, 2021-04-20:
I'm sorry but I'd like to ask you a question. Because I think I misunderstood me. But do I have to buy the unlock code from DC-Uncloker?

plarfman, 2021-04-20:
@MrBlood01 Nope, just a download link from DC unlocker servers, the driver itself has nothing to do with it

MrBlood01, 2021-04-20:
@plarfman So I just have to download the tool, install the drivers and open the tool.  But I don't have the unlock code

plarfman, 2021-04-21:
@MrBlood01 what error does it give? provide screenshot

MrBlood01, 2021-04-27:
@plarfman Maybe I'm wrong "PIN".  I can't figure out what the correct test point is on the motherboard of a Huawei p10 lite

MrBlood01, 2021-05-02:
@plarfman I can't figure out how to install COM drivers. I have by downloading the file from your link. But I can't install them. Here's what happens if I open the folder: \ Huawei drivers testpoint \ Driver \ X64

[image]

BudDeuce, 2021-05-02:
[image]

BudDeuce, 2021-05-02:
Run "DriverSetup.exe" as an administrator by right-clicking it and selecting "Run as administrator" and tell me the results.

MrBlood01, 2021-05-02:
So the test point is the one indicated by the red line or should I also press on the point indicated by the black zone?

Nothing has changed, I got this message. But it's the exact same message I got when installing it normally.

BudDeuce, 2021-05-02:
Try pressing down on just the yellow circle with the tweezers first.

MrBlood01, 2021-05-02:
I tried to take it at this point while plugging in the usb port.

[image]

MrBlood01, 2021-05-02:
@BudDeuce So now the drivers are installed correctly? Shouldn't it show me the COM port when I connect the device via cable?

BudDeuce, 2021-05-02:
Maybe you should press down the yellow circle and the area where the black line is pointing.

MrBlood01, 2021-05-02:
So am I supposed to create some kind of bridge between the two?

BudDeuce, 2021-05-02:
Yes, connect them at the same time using the tweezers.

MrBlood01, 2021-05-02:
I try immediately. But as I told you before, now I have connected my smartphone with active USB debus but I do not see the COM port

BudDeuce, 2021-05-02:
Send me a video of the process, please.

MrBlood01, 2021-05-02:
The process while installing the drivers?

BudDeuce, 2021-05-02:
Wait for the drivers to finish installing and then connect the phone.

It's the process where you use the tweezers and connect your phone at the same time.

MrBlood01, 2021-05-02:
I believe he has finished installing the drivers

[image]

BudDeuce, 2021-05-02:
Yep, I also believe the drivers are finished installing.

MrBlood01, 2021-05-02:
Perfect, I was thinking that I can record what happens on the screen. Because I can't use the phone while doing the operations. I need a third arm hahaha

BudDeuce, 2021-05-02:
I'm glad everything is going well for you too, if anything happens, just comment on this issue.

MrBlood01, 2021-05-02:
But have you succeeded? Does it work? 

Isn't there a risk of breaking the smartphone?

BudDeuce, 2021-05-02:
I did this with my Huawei P10 VTR-L09, everything went well.

As long as you don't touch any other components of the phone, I think you will be alright.

MrBlood01, 2021-05-02:
Perfect. So for the Huawei P10 Lite that gold dot is not enough?

BudDeuce, 2021-05-02:
I think that where the black line is pointing is to help keep the tweezers still.

BudDeuce, 2021-05-02:
I just touched the gold dot last time and that was enough for the operation to work.

MrBlood01, 2021-05-02:
okay, i try. Give me a second and above all thanks for the help

MrBlood01, 2021-05-02:
But my problem is that when I plug in the cable the software doesn't detect the device

BudDeuce, 2021-05-02:
Maybe reassemble your device and fully power it off again.

MrBlood01, 2021-05-02:
Okay, I'll try. After all, I only disconnected the battery

BudDeuce, 2021-05-02:
Don't disconnect the battery or else the device won't power on.

MrBlood01, 2021-05-02:
So the battery has to be connected? 
I was following a video on youtube he disconnected it while he breezed on the test point.

https://www.youtube.com/watch?v=YkGugQ019ZY&t=605s

MrBlood01, 2021-05-02:
In fact in this guide it does not say to disconnect the battery: https://github.com/mashed-potatoes/PotatoNV/wiki/How-to-use-PotatoNV

plarfman, 2021-05-02:
@MrBlood01 you don't have to unplug battery for testpoint, i can confirm on 2 devices

MrBlood01, 2021-05-02:
So that's why it doesn't work. Sorry I'm an idiot hahaha. I try

MrBlood01, 2021-05-02:
Thanks guys, you are great

BudDeuce, 2021-05-02:
No problem.

plarfman, 2021-05-02:
@MrBlood01 well it will work without battery too, it dosen't matter if its plugged in or not

MrBlood01, 2021-05-02:
Guys when I press the gold dot and shield on the motherboard the phone logs on and goes into this mode:

[image]

The computer doesn't detect anything.

MrBlood01, 2021-05-02:
After connecting the device I should see the COM port between my devices right?

BudDeuce, 2021-05-02:
Yes

BudDeuce, 2021-05-02:
Unplug all the other cables connecting to the motherboard where the test point is except the battery cable.

BudDeuce, 2021-05-02:
See if that works.

MrBlood01, 2021-05-02:
But now I also see this device:

[image]

BudDeuce, 2021-05-02:
What do you see in PotatoNV?

MrBlood01, 2021-05-02:
Can I start it?

BudDeuce, 2021-05-02:
Sure

MrBlood01, 2021-05-02:
Nothing for now

[image]

BudDeuce, 2021-05-02:
Try the latest version of PotatoNV here https://github.com/mashed-potatoes/PotatoNV/releases/download/v2.2.0/PotatoNV-next-v2.2.0-x86.exe

MrBlood01, 2021-05-02:
Sure, give me second

MrBlood01, 2021-05-02:
[image]

Nothing, in my opinion there is some problem with the drivers. But did you have to manually install any drivers?

plarfman, 2021-05-02:
@MrBlood01 you have somehow booted erecovery mode, are you sure you didn't hold volume up button or something?

MrBlood01, 2021-05-02:
Yes I'm sure. I pressed on the gold dot and on the motherboard. Immediately after that I connected the cable.

I can turn it off and try again if you want

BudDeuce, 2021-05-02:
Sure.

BudDeuce, 2021-05-02:
Do you have ADB installed?

[45 more replies]

closed 2021-05-02 by plarfman