kitsuned/PotatoNV #60: Auto Re-Locked

AhsanBulbul00, 2021-04-25:
Device: Huawei GR 5 2017 (BLL-L22) [ Kirin65(A) ]

When went for flashing twrp after successful unlocking,
`fastboot: FAILED (remote: image verification error) `
```
fastboot oem get-lockstate 
Locked!
fastboot oem get-bootinfo
Unlocked!
```
Then it started. OEM unlock in dev option greyed out.... And not fully unlocked or either locked. Any idea what to do next?
My end goal is to root that device....

Help would be appreciated.

plarfman, 2021-04-25:
on EMUI 5 or lower, you have to unlock with the code you set to on PotatoNV via fastboot 
fastboot oem unlock (codeyousetto)

mashed-potatoes, 2021-04-25:
@plarfman test test test

AhsanBulbul00, 2021-04-25:
Thanks man! Seems to unlock bootloader. But not booting into twrp though...

plarfman, 2021-04-25:
@AhsanBulbul00 what TWRP are you using? what happens when you try to boot it?

AhsanBulbul00, 2021-04-25:
Trying to flash twrp, as there isn't prob. other custom recovery for the device. After flashing, when I reboot, takes me straight to normal reboot page.

TWRP official site says to manually go to recovery when `fastboot reboot` ing as, Huawei recovery may auto replace any custom rec.
But not working for me...

closed 2021-04-26 by AhsanBulbul00

hleclere, 2021-05-06:
I am facing the same issue with Honor 6x unlocked with PotatoNV. The phone is running under EMUI 5.

@AhsanBulbul00 are you saying that using fastboot oem unlock (codeyousetto) did not solve the problem ?

Cheers