kitsuned/PotatoNV #61: will this work for honor 8 lite pra al00x ?

amrithmmh, 2021-04-27:
Hi, 
I am desperate to get the unlock code to root my device. can you add support for hoor 8 lite ra-al00x
HiSilicon Kirin 655
will this tool work?

mashed-potatoes, 2021-04-27:
@amrithmmh try to read

amrithmmh, 2021-04-27:
Hi,
I dont want to brick my device. I can see the tool suuports 65X series processor.Thanks for making such an awesome tool!

mashed-potatoes, 2021-04-27:
@amrithmmh https://github.com/mashed-potatoes/PotatoNV/blob/master/README.md#tested-devices

Huawei P8 Lite (2017) | PRA | Kirin 65x (A)

amrithmmh, 2021-04-27:
@mashed-potatoes  The device Huawei P8 lite is a different device.
I have Huawei Honor 8 lite - PRA-AL00x
You can add Honor 8 lite - PRA-AL00x to the list of tested devices.
Thanks a lot!

I just tried it anyway and it worked.
For anyone doing this:
read: https://github.com/mashed-potatoes/PotatoNV/wiki/How-to-use-PotatoNV
1. select kirin 65X(A) 
2. enjoy

It works 100% .

closed 2021-04-27 by amrithmmh

mashed-potatoes, 2021-04-27:
@amrithmmh The brand doesn't matter.
There are no differences (excl. NFC).