kitsuned/PotatoNV #66: Huawei P10 Lite: Failed to set FBLOCK state!

AceKiron, 2021-05-13:
I have a Huawei P10 Lite WAS-LX1A and want to use PotatoNV to set a new bootloader unlock code so I can unlock it.
However, it fails to set FBLOCK state? Here's a screenshot of PotatoNV.
[image]
On my phone, I'm in fastboot and it says PHONE Locked and FRP Unlock.

plarfman, 2021-05-13:
@AceKiron read the instructions.

BudDeuce, 2021-05-13:
You have to be in test point mode in order for the program to work.

BudDeuce, 2021-05-13:
[image]

AceKiron, 2021-05-13:
In that case, is there a way to be in testpoint mode without opening up my phone physically?

plarfman, 2021-05-13:
@AceKiron Nope, you have to open it

BudDeuce, 2021-05-13:
You could pay for something like HCU-Client or DC Unlocker, but it costs money.

AceKiron, 2021-05-13:
@plarfman is there at least a way without the hair dryer then? I feel like I'd accidentally overheat my phone if I use that.

BudDeuce, 2021-05-13:
@AceKiron Does your hair dryer support 60 degrees celsius?

AceKiron, 2021-05-13:
idk actually, and I can't ask my parents without them questioning why I want to know...

BudDeuce, 2021-05-13:
Are you trying to disassemble your parent's phone or your phone?

AceKiron, 2021-05-13:
my phone

BudDeuce, 2021-05-13:
It looks like you'll need convincing to perform the operation.

AceKiron, 2021-05-13:
yeah :/

BudDeuce, 2021-05-13:
Good luck.

AceKiron, 2021-05-13:
thanks, I'm gonna need it