kitsuned/PotatoNV #67: Huawei Y6 II - CAM-l21 COM driver issue

SoiThe, 2021-05-23:
Hello I have a problem with COM Drivers. 
[image]
This is what i get after install drivers from readme. I don't get any driver like "Huawei USB COM 1.0". So i can't choose anything in "target device" in PotatoNV. What drivers should I install?

BudDeuce, 2021-05-23:
Where did you get your COM drivers from?

SoiThe, 2021-05-23:
From here: https://github.com/mashed-potatoes/PotatoNV/wiki/How-to-use-PotatoNV

BudDeuce, 2021-05-23:
Did you have the device in test point mode before you installed the drivers?

SoiThe, 2021-05-23:
Ok, I fix the driver. But now I have a problem with PotatoNV. After click "start!" icon i get this:
Verifying images...
Uploading hisi620...
- xloader
- fastboot
The write timed out.

BudDeuce, 2021-05-23:
Do you have ADB installed?

SoiThe, 2021-05-23:
yes, from there: https://adb.clockworkmod.com/

BudDeuce, 2021-05-23:
Which version of PotatoNV do you have?

SoiThe, 2021-05-23:
v2.1.0

SoiThe, 2021-05-23:
When my phone is normally connected to the pc i get this drivers:
[image]

BudDeuce, 2021-05-23:
It's the correct drivers

SoiThe, 2021-05-23:
So what's the problem? My phone just can't be unlocked with this tool? I don't know what I could do with this "The write timed out" message.

2021-05-24 01:07:54.235 | INFO     | PotatoNV v2.1.0
2021-05-24 01:07:54.249 | INFO     | User manual: https://kutt.it/pnv-en
2021-05-24 01:08:11.664 | DEBUG    | DownloadVCOM mode: COM7: HUAWEI USB COM 1.0
2021-05-24 01:08:19.532 | INFO     | PotatoNV v2.1.0
2021-05-24 01:08:19.532 | INFO     | User manual: https://kutt.it/pnv-en
2021-05-24 01:08:19.534 | INFO     | Verifying images...
2021-05-24 01:08:19.542 | DEBUG    | Opening COM7...
2021-05-24 01:08:19.548 | INFO     | Uploading hisi620...
2021-05-24 01:08:19.549 | INFO     | - xloader
2021-05-24 01:08:20.707 | INFO     | - fastboot
2021-05-24 01:08:21.708 | ERROR    | The write timed out.
2021-05-24 01:08:21.719 | DEBUG    | at System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
    at System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
    at Potato.ImageFlasher.ImageFlasher.SendFrame(Byte[] data)
    at Potato.ImageFlasher.ImageFlasher.SendHeadFrame(Int32 length, Int32 address)
    at Potato.ImageFlasher.ImageFlasher.Write(String path, Int32 address, Action`1 reportProgress)
    at PotatoNV_next.Core.FlashBootloader(Bootloader bootloader, String port)
    at PotatoNV_next.Core.Worker_DoWork(Object sender, DoWorkEventArgs e)
2021-05-24 01:08:39.898 | DEBUG    | DownloadVCOM mode: COM7: HUAWEI USB COM 1.0

That's what i get in the logs file.

BudDeuce, 2021-05-23:
Are you using a USB 3.0 port?

SoiThe, 2021-05-23:
Nope, I use 2.0

BudDeuce, 2021-05-23:
No USB hubs?

SoiThe, 2021-05-23:
nope

BudDeuce, 2021-05-23:
Try going a version back, e.x. V 2.0.0

SoiThe, 2021-05-24:
Do you know where can i get thi 2.0.0 version? I haven't seen it here on github

mashed-potatoes, 2021-05-24:
@SoiThe Changing version won't make any difference. Your device is running/charging. Check the wire/tweezers as well as the testpoint.

SoiThe, 2021-05-24:
Yeah, this phone have kirin 620, so i can only use the newest version. After this "the write timed out" message, phone diode is blinking in two colours - green and red

mashed-potatoes, 2021-05-24:
@SoiThe It looks like the bootloader is incompatible with CAM. I'll send you alternative bootloader soon.
Stay tuned.

SoiThe, 2021-05-24:
Okay

seisdr, 2021-06-04:
iI also have the same device if you managed to rooted it or unlock its bootloader please let me know

seisdr, 2021-06-04:
pictures of how you connect the cables to the testpoint would be helpful

insking98, 2021-06-09:
Same problem here "write timed out" for y62 CAM l21

seisdr, 2021-06-17:
what happen to the alternative bootloader

BudDeuce, 2021-06-17:
Exactly.

seisdr, 2021-06-17:
I'm patient

seisdr, 2021-07-10:
hello again did anyone mange to unlock the device bootloader? I haven't tried yet

BudDeuce, 2021-07-10:
If you want to unlock today, you can buy HCU-Client, which is paid. Or you can wait for who knows how long for Andrey to make a compatible bootloader.

seisdr, 2021-07-10:
I'm saying because there's update released 12 days ago maybe will work , I wait

plarfman, 2021-07-11:
@BudDeuce the 2.2.1 version has the CAM bootloader already

SoiThe, 2021-07-11:
> hello again did anyone mange to unlock the device bootloader? I haven't tried yet

@seisdr Yes, I unlock my cam-I21. In my case this software only change the bootloader code. So I have to copy that code and unlock my phone using mini adb and fastboot.

BudDeuce, 2021-07-11:
@plarfman I see.

seisdr, 2021-07-12:
@SoiThe excuse me how did you fix the driver issue?

SoiThe, 2021-07-12:
@seisdr I wired wrong pin to enter testpoint. I watched how to use this software on youtube and in this video a guy have different smartphone. He wired testpoint and metal cover on motherboard. In my phone i have to wire two points on motherboard to enter testpoint.
In my case that was that two dots.
[image]

seisdr, 2021-07-12:
thanks I was doing it wrong , but the thing is the potatonv doesn't pick up the phone when it's turned off  not in testpoint mode,  do I have to enter testpoint mode so it detect the phone? or there is issue with the testpoint drivers on windows ?

SoiThe, 2021-07-12:
@seisdr If you install adb drivers and com drivers you have to shutdown your phone, wire testpoint pins (i used tweezers) and then connect cable to the phone. You have to wired testpoint pins all the time. Before that you can open potatonv or device manager to see that your phone is connected properly.
This video helped me: https://www.youtube.com/watch?v=YkGugQ019ZY

seisdr, 2021-07-12:
It worked I finally unlock it ! Thank you @SoiThe
Now how did you root it ? I don't want to bricke it as I just

closed 2021-08-01 by mashed-potatoes