kitsuned/PotatoNV #68: P10 device and FRP unlocked, but device in fastboot/bootloader mode not detected

antivirtel, 2021-05-29:
Hi,

I've successfully unlocked the device (P10 VTR-L09) - shows the unlock warning, shows device unlock, thank you! However, then tried to flash TWRP, but when I get into the bootloader (aften an `adb reboot bootloader` or Volume Down + plug in USB) the device doesn't come up neither in Windows 10 nor in Linux at all, not even unrecognised device, nothing. I've tried 3 different cables, and 2 different computers (Win10 and Linux). When I plug this phone in in bootloader mode into Linux, it doesn't even show up in `dmesg` output. In normal boot, I can easily play around with `adb`, that works fine in both computers, so I doubt there is anything wrong with either the cables or the computers.

This is how it looks like in bootloader mode:

What can I do to let me flash TWRP?

Thanks!

BudDeuce, 2021-05-29:
Which command are you using to flash TWRP?

antivirtel, 2021-05-29:
`fastboot flash recovery twrp.img` - just tells me that it is waiting for a device, nothing happens.

BudDeuce, 2021-05-29:
It's "fastboot flash recovery_ramdisk twrp.img"

antivirtel, 2021-05-29:
Tried it still the same:
```
> fastboot flash recovery_ramdisk twrp-3.1.1-0-vtr.img
< waiting for any device >
```

BudDeuce, 2021-05-29:
Try disabling Driver signature enforcement through this link: https://windowsreport.com/driver-signature-enforcement-windows-10/#:~:text=Press%20and%20hold%20the%20Shift,select%20Disable%20driver%20signature%20enforcement.

antivirtel, 2021-05-29:
Tried that, no difference.

BudDeuce, 2021-05-29:
Where did you get your drivers from?

antivirtel, 2021-05-29:
From the links here: https://github.com/mashed-potatoes/PotatoNV/wiki/How-to-use-PotatoNV - I don't think it gets to the point when I need the drivers, as it literally nothing changes in device manager in Windows, and nothing in `dmesg`/`lsusb` on Linux.

BudDeuce, 2021-05-29:
Is there a yellow exclamation mark triangle in device manager?

antivirtel, 2021-05-29:
No, all ok, and there's no new device when I want to plug it in, I think there's the problem...

BudDeuce, 2021-05-29:
Right-click the bootloader interface (if there is one) and try to find an error code in "General".

antivirtel, 2021-05-29:
There is no device in device manager...

BudDeuce, 2021-05-29:
Are you using a USB 2.0 port?

antivirtel, 2021-05-29:
I have both, tried them both, same.

BudDeuce, 2021-05-29:
Can you send a photo of your device manager window?

antivirtel, 2021-05-29:
Sure
[image]

BudDeuce, 2021-05-29:
What is your computer model? You could have outdated USB drivers.

antivirtel, 2021-05-29:
HP Z420

BudDeuce, 2021-05-29:
Update all the USB-related drivers by right-clicking on them and pressing "Update Driver" then click "Search automatically for drivers."

antivirtel, 2021-05-29:
Did it, it said already have the best

BudDeuce, 2021-05-29:
Try restarting your computer.

antivirtel, 2021-05-29:
I did that multiple times

BudDeuce, 2021-05-29:
And your phone?

antivirtel, 2021-05-29:
Oh yeah, like 10 times already. Always works in ADB when booting normally, nothing in fastboot

BudDeuce, 2021-05-29:
Try installing HiSuite, which comes with the newest Huawei drivers.

BudDeuce, 2021-05-29:
Any luck so far?

mashed-potatoes, 2021-05-30:
@BudDeuce there is no point in reinstalling drivers if the computer is unable to detect the device.

BudDeuce, 2021-05-30:
I know, but there might be something involving the fastboot part of the drivers. Anything is possible.

mashed-potatoes, 2021-05-30:
@antivirtel try this:
1. connect your phone in testpoint mode
2. close the program as soon as you see "waiting for any device..."
3. the device should switch to fastboot

antivirtel, 2021-05-30:
Thank you for the replies. @mashed-potatoes - about the 2nd step, which program should I close? Try to do the `fastboot flash ...` then `Ctrl` + `C` it when `< waiting for any device >` is visible?

mashed-potatoes, 2021-05-30:
@antivirtel you should close PotatoNV.

antivirtel, 2021-06-03:
@mashed-potatoes thanks, it takes me to fastboot mode, I have it as Android Phone > Android Bootloader Interface (or 2 other with ADB, but same result), I tried from Windows Admin PS, CMD and Linux, same on all:
```
PS C:\tmp\Huawei> fastboot devices
6PQ0217B03001347        fastboot

PS C:\tmp\Huawei> fastboot flash recovery_ramdisk twrp-3.1.1-0-vtr.img
Sending 'recovery_ramdisk' (32040 KB)              OKAY [  0.959s]
Writing 'recovery_ramdisk'                         FAILED (remote: 'partition length get error')
fastboot: error: Command failed

PS C:\tmp\Huawei> fastboot flash recovery twrp-3.1.1-0-vtr.img
Sending 'recovery' (32040 KB)                      OKAY [  0.960s]
Writing 'recovery'                                 FAILED (remote: 'image verification error')
fastboot: error: Command failed

PS C:\tmp\Huawei> fastboot boot twrp-3.1.1-0-vtr.img
Sending 'boot.img' (32040 KB)                      OKAY [  1.023s]
Booting                                            FAILED (remote: 'invalid command')
fastboot: error: Command failed
```

I've tried this: https://forum.xda-developers.com/t/recovery-8-0-unofficial-twrp-3-2-1-0-for-emui-8-0-06-05-2018.3773449/ https://huaweiadvices.com/huawei-p10-install-twrp-recovery/ https://clickitornot.com/install-twrp-recovery-and-root-huawei-p10/
Any ideas why it can't take the image?

mashed-potatoes, 2021-06-03:
@antivirtel Use fastboot oem unlock [CODE]

antivirtel, 2021-06-03:
Thanks, that helped, but the recovery can't seem to work, I can't flash it for `recovery_ramdisk`, and on `recovery` it just doesn't work. Do you have any idea why? I tried all 3 TWRP images linked above.

[image]

```
PS C:\tmp\Huawei> fastboot flash recovery_ramdisk .\TWRP_P10_Lite_Beta.img
Sending 'recovery_ramdisk' (24468 KB)              OKAY [  0.728s]
Writing 'recovery_ramdisk'                         FAILED (remote: 'partition length get error')
fastboot: error: Command failed

PS C:\tmp\Huawei> fastboot flash recovery .\TWRP_P10_Lite_Beta.img
Sending 'recovery' (24468 KB)                      OKAY [  0.804s]
Writing 'recovery'                                 OKAY [  0.148s]
Finished. Total time: 0.984s

PS C:\tmp\Huawei> fastboot boot TWRP_P10_Lite_Beta.img
Sending 'boot.img' (24468 KB)                      OKAY [  0.785s]
Booting                                            FAILED (remote: 'invalid command')
fastboot: error: Command failed

PS C:\tmp\Huawei> fastboot reboot recovery
Rebooting into recovery                            OKAY [  0.013s]
Finished. Total time: 0.015s
```

BudDeuce, 2021-06-04:
@antivirtel Did you try Pretoriano80's new version of TWRP? If you didn't, here's the link: https://mega.nz/file/9xVgxDwT#NOk3XkSaBX8PC5xjyrZNyIIz52aSoPZFphQNauhTKGQ

antivirtel, 2021-06-04:
Yeah, it's liked here, I've linked it before: https://forum.xda-developers.com/t/recovery-8-0-unofficial-twrp-3-2-1-0-for-emui-8-0-06-05-2018.3773449/ - same error, it's probably not the image then if none of them work. Can be some storage or partitioning issue?

BudDeuce, 2021-06-04:
What are you trying to do with TWRP?

antivirtel, 2021-06-04:
The XDA post says it's for EMUI 8, but I have EMUI 9, can it be an issue?
[image]

I'm trying to do the usual, TWRP to flash zip files, flash in a good Custom ROM, eg. LineageOS, but if it is full of bugs, not maintained either, just restore the Stock ROM, then root with Magisk or SuperSU... Probably can do it without TWRP, but more pain, isn't it? Mainly if I need to play around with the test point every time before entering fastboot mode.

BudDeuce, 2021-06-04:
That could be a problem, every phone is different.

antivirtel, 2021-06-04:
That's sad, @mashed-potatoes any more ideas?

mashed-potatoes, 2021-06-11:
> That's sad, @mashed-potatoes any more ideas?

try TWRPs for STF.
or use it: [victoria_fp_gore.img.gz](https://github.com/mashed-potatoes/PotatoNV/files/6640955/victoria_fp_gore.img.gz).

antivirtel, 2021-06-13:
I've tried this 4, no luck, same issue. Any idea why?

- yours
- https://ministryofsolutions.com/2020/01/honor-9-stf-l09-twrp-and-root-emui9-1.html
- https://forum.xda-developers.com/t/stf_emui5-1_twrp-v2-with-dual-recovery-support-stock-and-twrp.3634637/
- https://forum.xda-developers.com/t/oreo-twrp-3-2-1-0-0-3-capacitive-fix.3754483/ 

```powershell
PS C:\tmp\Huawei> fastboot flash bootloader_ramdisk .\stf-twrp.img
Sending 'bootloader_ramdisk' (33798 KB)            OKAY [  1.005s]
Writing 'bootloader_ramdisk'                       FAILED (remote: 'partition length get error')
fastboot: error: Command failed

PS C:\tmp\Huawei> fastboot flash bootloader_ramdisk TWRP.img
Sending 'bootloader_ramdisk' (26448 KB)            OKAY [  0.787s]
Writing 'bootloader_ramdisk'                       FAILED (remote: 'partition length get error')
fastboot: error: Command failed

PS C:\tmp\Huawei> fastboot flash bootloader_ramdisk .\victoria_fp_gore.img
Sending 'bootloader_ramdisk' (25304 KB)            OKAY [  0.749s]
Writing 'bootloader_ramdisk'                       FAILED (remote: 'partition length get error')
fastboot: error: Command failed

PS C:\tmp\Huawei> fastboot flash bootloader_ramdisk .\stf-twrp-zxz-0.3.img
Sending 'bootloader_ramdisk' (27144 KB)            OKAY [  0.809s]
Writing 'bootloader_ramdisk'                       FAILED (remote: 'partition length get error')
```

BudDeuce, 2021-06-13:
It's "recovery_ramdisk" not bootloader_ramdisk.

antivirtel, 2021-06-13:
Oh, my bad. But it changes nothing.
```console
PS C:\tmp\Huawei> fastboot flash recovery_ramdisk TWRP.img
Sending 'recovery_ramdisk' (26448 KB)              OKAY [  0.790s]
Writing 'recovery_ramdisk'                         FAILED (remote: 'partition length get error')
fastboot: error: Command failed
```

I can do the `recovery` partition, but I can't boot into that, see the error above.
```console
PS C:\tmp\Huawei> fastboot flash recovery TWRP.img
Sending 'recovery' (26448 KB)                      OKAY [  0.765s]
Writing 'recovery'                                 OKAY [  0.141s]
Finished. Total time: 0.930s
```

In the meantime, I've found some interesting stuff from Magisk: https://github.com/topjohnwu/Magisk/blob/408399eae095b7cbd3e05278682c4bb4c7702ec0/docs/install.md#huawei - I've checked with the Magisk app, it says ramdisk: No - screenshot:
[image]

> After downloading your firmware zip (you may find a lot in [Huawei Firmware Database](http://pro-teammt.ru/firmware-database/)), you have to extract images from UPDATE.APP in the zip with [Huawei Update Extractor](https://forum.xda-developers.com/showthread.php?t=2433454) (Windows only!)
> - If your device does NOT have boot ramdisk, patch `RECOVERY_RAMDIS.img` (this is not a typo) instead of `recovery.img`
> - If you patched `RECOVERY_RAMDIS.img`, flash with command `fastboot flash recovery_ramdisk magisk_patched.img`

Problem is I can't find the FW, all the sites are 404. Eg. http://update.hicloud.com:8180/TDS/data/files/p3/s15/G2323/g1699/v127945/f1/full/update.zip - is there a way of getting this from the device?

BudDeuce, 2021-06-14:
https://github.com/ProfessorJTJ/HISuite-Proxy/wiki/Complete-Guide

antivirtel, 2021-06-20:
Thanks BudDeuce, I've tried it, 2 things:

1. It says in HiSuite when I press Update that "You're now using the latest official version". But where are the images? I don't want to upgrade the phone, I just need the images.
[image]

2. It's not a full match on firmfinder.ml, this is what I find:
284018 2019/08/14 | VTR-L09 - C440 | VTR-L09 9.0.1.189(C440E3R1P8)
but I have this:
[image]

Is it an issue or should I try to find another site?

Thanks

antivirtel, 2021-07-15:
Hi, Do you have any other suggestions? Thanks

BudDeuce, 2021-07-15:
I don't have any other suggestions.