kitsuned/PotatoNV #71: how to compile that lol

zielony12, 2021-06-08:
idk how do i compile it plz help xd

newbit1, 2021-06-08:
I did it with the [Free Edition of Visual Studio](https://visualstudio.microsoft.com/downloads/).
After Downloading and Installing, I've just added the git repo to it, and build it.
Make sure git is in your path, otherwise you will get an error with "git submodule" from
the C# Project file [PotatoNV-next.csproj](https://github.com/mashed-potatoes/PotatoNV/blob/0de740364003d16c7e0520b9814f1565e8dc7e66/PotatoNV-next/PotatoNV-next.csproj#L206)

zielony12, 2021-06-10:
> I did it with the [Free Edition of Visual Studio](https://visualstudio.microsoft.com/downloads/).
> After Downloading and Installing, I've just added the git repo to it, and build it.
> Make sure git is [...]

lol i use arch btw

mashed-potatoes, 2021-06-11:
https://github.com/mashed-potatoes/PotatoNV-crossplatform
have fun.

closed 2021-06-11 by mashed-potatoes