kitsuned/PotatoNV #72: Huawei P9 partition length get error when trying to flash TWRP

DJGummikuh, 2021-06-11:
Hey! I successfully unlocked my phone but now lack the ability to flash twrp because (as far as I understood) my phone is on Android 8 and does not accept the TWRP versions for the P9 (the latest of which was released in 2017. Is there any newer version I can use?

DJGummikuh, 2021-06-11:
I'm using fastboot flash recovery twrp-3.1.1-0-eva.img (from the twrp download page). Also tried 3.1.0-0 and 3.0.2-0, all with the same response.

I believe the ramdisk stuff came with oreo, but the TWRP versions seem to be too old for that

DJGummikuh, 2021-06-11:
(to be fair, I'm actually surprised my P9 even got the Android 8 update)

BudDeuce, 2021-06-11:
Try "fastboot flash recovery_ramdisk twrp-3.1.1-0-eva.img"

DJGummikuh, 2021-06-11:
I officially like to let you know that I'm an idiot. I always read the ramdisk part as part of the filename, not the command >.<

Thanks for helping me out!

BudDeuce, 2021-06-11:
No problem.

DJGummikuh, 2021-06-11:
I need your help once again, I flashed into recovery_ramdisk but when I adb reboot into recovery I get an error mmode "Attention! please update system again! Error! Func NO: 11 (recovery image)
Error NO: 2 (load failed!)
Phone still can boot normally into android but I can't access TWRP like that

BudDeuce, 2021-06-11:
Which version of TWRP did you flash?

DJGummikuh, 2021-06-11:
3.1.1.0
this is a bit of a hassle. This guide here says "3.4.0 or later": https://www.getdroidtips.com/root-install-twrp-recovery-huawei-p9/
but the link then points to a download page where I can download 3.1.1.0 as the latest.

BudDeuce, 2021-06-11:
Try going back 1 version, see what happens.

DJGummikuh, 2021-06-11:
one version down didnt help, trying the oldest one now (there were 3 files offered)

DJGummikuh, 2021-06-11:
nope, all 3 offered twrps runninto the same error mode.

BudDeuce, 2021-06-11:
https://www26.zippyshare.com/v/JI2h29gy/file.html Try the TWRP included in the data folder of this file.

DJGummikuh, 2021-06-11:
I get a 403 forbidden o.O

BudDeuce, 2021-06-11:
Weird.

DJGummikuh, 2021-06-11:
[image]

BudDeuce, 2021-06-11:
https://www26.zippyshare.com/d/JI2h29gy/37226/ROOT_HUAWEI_P9.zip Here is the direct download link.

DJGummikuh, 2021-06-11:
[image]
Am I stupid or something??

BudDeuce, 2021-06-11:
No, the TWRP may be EMUI version specific.

DJGummikuh, 2021-06-11:
flashed the twrp.img from that download file, same result. Func NO:11 Err NO: 2

BudDeuce, 2021-06-11:
Maybe your EMUI is not compatible with TWRP at the moment.

DJGummikuh, 2021-06-11:
I mean I want to remove all that and install lineage OS anyways. 
Model: EVA-AL00
Build-Number: EVA-AL00 8.0.0.552(C00)
EMUI-Version: 8.0.0
Android-Version: 8.0.0
Is there a list anywhere that helps me know if TWRP is compatible with that and - if so - which version to get where?

BudDeuce, 2021-06-11:
https://androidfilehost.com/?fid=1322778262904008816 I finally found a TWRP that is compatible with your EMUI version.

BudDeuce, 2021-06-11:
https://androidfilehost.com/?fid=1322778262903998750 Sorry, that was for the P9 PLUS, not P9

DJGummikuh, 2021-06-11:
Where are you from anyways? If we ever cross paths I am SO going to buy you a beer :-D

BudDeuce, 2021-06-11:
I want to keep my location a secret, but thanks for the complement.

DJGummikuh, 2021-06-11:
Fair enough

DJGummikuh, 2021-06-11:
booted into recovery now! you are my hero of the day!

closed 2021-06-11 by mashed-potatoes

DJGummikuh, 2021-06-11:
Ok now after flashing lineageOS I entered a boot loop. Have to continue tomorrow