kitsuned/PotatoNV #74: Huawei P8 Lite Alice ALE-L21 2015 rel

lm212121, 2021-07-08:
hello

I have this model, as alice l21 from 2015 hw release. Lot of LOS, twrp or others freedom-oss-like topics on xda regarding p8 lite are about the 2017 rel. I do have the 2015 only.
well, I tried with the potatonv tool to unlock the phone. The operation looks like to success, but unfortunately, even if the screen shows that the bootloader is unlocked.... 

pc ~ # fastboot oem get-lockstate
...
(bootloader) locked
OKAY [  0.001s]
finished. total time: 0.001s
pc ~ # fastboot oem get-bootinfo
...
(bootloader)  unlocked
OKAY [  0.000s]
finished. total time: 0.000s
pc ~ # fastboot flash recovery /tmp/twrp_alice21/recovery.img 
target reported max download size of 471859200 bytes
sending 'recovery' (25152 KB)...
OKAY [  0.627s]
writing 'recovery'...
FAILED (remote: image verification error)
finished. total time: 0.632s

what could I do?

is it possible to communicate with the tool creator Andrey Smirnoff by email, irc or other forum/tchat subject?

I thank you vm

BudDeuce, 2021-07-08:
Try "recovery_ramdisk" instead of "recovery".

lm212121, 2021-07-08:
> Try "recovery_ramdisk" instead of "recovery".

fastboot flash recovery_ramdisk /tmp/twrp_alice21/recovery.img 
target reported max download size of 471859200 bytes
sending 'recovery_ramdisk' (25152 KB)...
OKAY [  0.617s]
writing 'recovery_ramdisk'...
FAILED (remote: partition does not exist)
finished. total time: 0.618s

BudDeuce, 2021-07-08:
Use the bootloader code that PotatoNV gave you and say "fastboot oem unlock bootloadercode"

lm212121, 2021-07-08:
> Use the bootloader code that PotatoNV gave you and say "fastboot oem unlock bootloadercode"

already tried a dozen of times

BudDeuce, 2021-07-08:
Did you check the FBLOCK option in PotatoNV?

lm212121, 2021-07-08:
> Did you check the FBLOCK option in PotatoNV?

yes

BudDeuce, 2021-07-08:
Which version of PotatoNV did you use?

lm212121, 2021-07-08:
> Which version of PotatoNV did you use?

2.1.0

BudDeuce, 2021-07-08:
If you go to the downloads page, there are new versions that contain bug fixes, try those.

lm212121, 2021-07-08:
> If you go to the downloads page, there are new versions that contain bug fixes, try those.

done today, still same issue : 
 fastboot flash recovery_ramdisk /tmp/twrp_alice21/recovery.img 
target reported max download size of 471859200 bytes
sending 'recovery_ramdisk' (25152 KB)...
OKAY [  0.606s]
writing 'recovery_ramdisk'...
FAILED (remote: partition does not exist)
finished. total time: 0.607s
pc ~ # fastboot flash recovery /tmp/twrp_alice21/recovery.img 
target reported max download size of 471859200 bytes
sending 'recovery' (25152 KB)...
OKAY [  0.594s]
writing 'recovery'...
FAILED (remote: image verification error)
finished. total time: 0.600s
pc ~ # fastboot oem get-bootinfo
...
(bootloader)  unlocked
OKAY [  0.001s]
finished. total time: 0.001s
pc ~ # fastboot oem get-lockstate
...
(bootloader) locked
OKAY [  0.001s]
finished. total time: 0.001s

lm212121, 2021-07-09:
is there a way to transmit this issue to the potato tool's developper?

lm212121, 2021-07-10:
.? :-/

BudDeuce, 2021-07-10:
Yes, there is. @mashed-potatoes

lm212121, 2021-07-11:
> Yes, there is. @mashed-potatoes

I mean without twitter nor social networks : email for example?
or on a forum?

lm212121, 2021-07-28:
hey guys, how could I solve the problem of huawei wrongly unlocked through potato?

lm212121, 2021-07-29:
looks like Im sending  a message in a bottle :
https://forum.xda-developers.com/t/twrp-openkirin-3-1-1-0-all-versions-huawei-p8lite.3583180/post-85398893

mashed-potatoes, 2021-08-01:
Disable FBLOCK to solve this problem:
```
fastboot oem backdoor set 1
```
Checkbox `Disable FBLOCK` will be disabled for legacy devices in the next release.

closed 2021-08-01 by mashed-potatoes