kitsuned/PotatoNV #78: P10 Lite: "The operation has timed out." (logs provided)

EmanueleCiriachi, 2021-07-25:
Connection is achieved as follows:
[image]

Yet this is the result:
[image]

Log is here:
[test.log](https://github.com/mashed-potatoes/PotatoNV/files/6874104/test.log)

What am I doing wrong?

mashed-potatoes, 2021-07-25:
@EmanueleCiriachi you're using wrong bootloader: https://github.com/mashed-potatoes/PotatoNV#tested-devices.

EmanueleCiriachi, 2021-07-25:
> @EmanueleCiriachi you're using wrong bootloader: https://github.com/mashed-potatoes/PotatoNV#tested-devices.

I tried all of them, it didn't work...

mashed-potatoes, 2021-07-25:
@EmanueleCiriachi Ah, I see. You should re-enter vcom mode after each attempt.

EmanueleCiriachi, 2021-07-25:
It worked! You're a literal savior. I'll go buy you a beer right now!

closed 2021-07-26 by EmanueleCiriachi

mashed-potatoes, 2021-07-26:
ily ❤️