kitsuned/PotatoNV #84: Huawei Mate 10 lite - Failed to set the FBLOCK

saayalac, 2021-08-08:
Hello, I am trying to unlock bootloader in order to fix a apparently soft-bricked huawei mate 10 lite

I am doing the process as descrived on the video. 

I use the test point to get the phone on that mode, driver installed first time, used both kirin 65x A and B option, used with and without "Disable FBLOCK" check option, always the same result

The phone gets to bootloader and says "PHONE unlocked" and "FRP Lock", but look at the logs.
Help :c

"PotatoNV v2.2.1
User manual: https://kutt.it/pnv-en
Verifying images...
Uploading hisi65x_a...
- xloader
- fastboot
Waiting for any device...
Connecting...
Serial number: M4VDU18317004776
Board ID: DU4RTR183F095200
Model: RNE-L23
Build number: RNE-L23 8.0.0.366(C605)
FBLOCK state: unlocked
Saved key: UUUUUUUUUUUUUUUU
Writing FBLOCK...
Failed to set the FBLOCK, using the alternative method...
Trying hwdog certify set...
Trying backdoor set...
Failed to set FBLOCK state!
Writing WVLOCK...
Failed to set the key.
New unlock code: M7QP4TGG2XTWO0Y5
"

[image]

BudDeuce, 2021-08-08:
Are you using fastboot or the test point method?

saayalac, 2021-08-08:
I use the test point method, it starts as expected in the test point mdoe, the app recognized the phone, I click on "start" it waits to  a green bar to fill up, then the phone restarts on fastboot and the app attempts to do the rest then there appers the errors.

saayalac, 2021-08-09:
help pls :c

BudDeuce, 2021-08-09:
I do not have any other explanation, sorry man.