kitsuned/PotatoNV #93: FBLock is Locked

ImAndrew0, 2021-09-01:
Hi, I have a Huawei P8 Lite 2017 Dual-Sim, I would like to unlock the bootloader with PotatoNV but when I click start it gives me an error: FBLock is Locked!

I have tried both the zip and the .exe

How can I solve?

BudDeuce, 2021-09-01:
Are you using fastboot or test point mode?

ImAndrew0, 2021-09-01:
Fastboot mode

BudDeuce, 2021-09-01:
It only works in test point mode. What is your model number?

ImAndrew0, 2021-09-01:
How to known?

BudDeuce, 2021-09-01:
Yep. https://www.youtube.com/watch?v=YkGugQ019ZY

BudDeuce, 2021-09-01:
It is required to disassemble the phone, yes.

ImAndrew0, 2021-09-01:
Ok thanks

BudDeuce, 2021-09-01:
No problem.

ImAndrew0, 2021-09-02:
> Yep. https://www.youtube.com/watch?v=YkGugQ019ZY

i tried to follow the guide but there is no OEM unlock option

mmaret, 2022-02-14:
I can confirm on my P8 lite 2017 with OS version 8.0 that potatoNV using the test point allow me to get a unlock code.
To then use this code, I have to enable the "Developer Settings" of the Android OS (Tap 5 times on the Build number in the system settings), and select "OEM unlock".
I was then able to reboot in fastboot mode (power up + vol down or adb reboot bootloader) and enter the following commands:
fastboot oem unlock CODEFOUNDBYPOTATONV