kitsuned/PotatoNV #102: VTR-L09 (P10) Times Out

ecrax, 2021-11-08:
I don't know if this project is still being maintained, but maybe some have the same problem. I can connect my phone using the testpoint drivers and also start PotatoNV. But once it reaches the "Waiting for any device..." stage, I see in the Win Device Manager that the phone disconnects. Then Potato throws a timeout error.

[image]

```
2021-11-08 15:28:13.449 | INFO     | PotatoNV v2.2.1
2021-11-08 15:28:13.452 | INFO     | User manual: https://kutt.it/pnv-en
2021-11-08 15:28:13.769 | DEBUG    | DownloadVCOM mode: COM6: HUAWEI USB COM 1.0
2021-11-08 15:28:19.995 | INFO     | PotatoNV v2.2.1
2021-11-08 15:28:20.010 | INFO     | User manual: https://kutt.it/pnv-en
2021-11-08 15:28:20.014 | INFO     | Verifying images...
2021-11-08 15:28:20.027 | DEBUG    | Opening COM6...
2021-11-08 15:28:20.030 | INFO     | Uploading hisi960...
2021-11-08 15:28:20.030 | INFO     | - xloader
2021-11-08 15:28:20.360 | INFO     | - uce
2021-11-08 15:28:20.767 | INFO     | - fastboot
2021-11-08 15:28:36.094 | INFO     | Waiting for any device...
2021-11-08 15:29:01.951 | ERROR    | Timeout error.
2021-11-08 15:29:01.956 | DEBUG    | at Potato.Fastboot.Fastboot.Wait()
    at PotatoNV_next.Core.Worker_DoWork(Object sender, DoWorkEventArgs e)
```

Any help is appreciated and if anyone needs more information, feel free to ask. :)

mashed-potatoes, 2021-11-08:
@ecrax try to install HiSuite and Google USB Driver

ecrax, 2021-11-08:
I have both installed
[image]
[image]

mashed-potatoes, 2021-11-08:
@ecrax do you see any "Unrecognized device" in device manager?

mashed-potatoes, 2021-11-08:
@ecrax start PotatoNV again, then check device manager again

ecrax, 2021-11-08:
It seems like it gets completely disconnected, there is no unusual device in the device manager

mashed-potatoes, 2021-11-08:
try to use USB 2.0 port or USB hub

mashed-potatoes, 2021-11-08:
try USB 2.0 port / USB hub

ecrax, 2021-11-08:
using a usb 2.0 port at the back of my computer worked, thanks a lot for your help and thanks for your work on PotatoNV :)
have a great day

closed 2021-11-08 by ecrax

mashed-potatoes, 2021-11-08:
good luck