kitsuned/PotatoNV #112: Honor P9 STF-L09. Timeout error.

dimashu, 2021-12-29:
Hi. Installed all drivers according to the instructions. Enabled test point. HUAWEI USB COM 1.0 appeared in the list of devices. But the program gives a Timeout error. What am I doing wrong?

2021-12-29 14:21:25.111 | INFO     | PotatoNV v2.2.1
2021-12-29 14:21:25.116 | INFO     | User manual: https://kutt.it/pnv-ru
2021-12-29 14:22:12.049 | DEBUG    | DownloadVCOM mode: COM4: HUAWEI USB COM 1.0
2021-12-29 14:22:23.548 | INFO     | PotatoNV v2.2.1
2021-12-29 14:22:23.550 | INFO     | User manual: https://kutt.it/pnv-ru
2021-12-29 14:22:23.555 | INFO     | Verifying images...
2021-12-29 14:22:23.567 | DEBUG    | Opening COM4...
2021-12-29 14:22:23.572 | INFO     | Uploading hisi960...
2021-12-29 14:22:23.572 | INFO     | - xloader
2021-12-29 14:22:23.881 | INFO     | - uce
2021-12-29 14:22:24.200 | INFO     | - fastboot
2021-12-29 14:22:38.881 | INFO     | Waiting for any device...
2021-12-29 14:23:04.539 | ERROR    | Timeout error.
2021-12-29 14:23:04.542 | DEBUG    | в Potato.Fastboot.Fastboot.Wait()
    в PotatoNV_next.Core.Worker_DoWork(Object sender, DoWorkEventArgs e)

plarfman, 2021-12-29:
@dimashu Install Fastboot & ADB drivers.

dimashu, 2021-12-29:
Yes, of course. Together with Android Studio.

mashed-potatoes, 2021-12-29:
@dimashu Install HiSuite and try again

dimashu, 2021-12-29:
HiSuite is installed too. The phone is full of bricks. Responds only to test point. Sometimes, when charging, after a while it displays the inscription "your device has failed verification and may not work properly". When the UP key is pressed, it tries to reboot but returns back.

bratekarate, 2022-01-08:
Same issue here. Driver install works without a problem, COM device is shown correctly in device manager and potatonv recognizes it. Chose Kirin 960, results in the same error as described above.

After the unlock fails, the COM device disappears from device manager. Instead, a device called hi3650 appears in device manager under "other devices" potatonv does not recognize it anymore. Only way for me to get it back is to turn the phone on and off again and short the testpoint again.

Any ideas?